hankedan000/megasquirt-dbw #29: MS3PRO

Ms3pro, 2023-05-16:
Hi! Does it work on ms3pro?

hankedan000, 2023-05-17:
This code is not designed to run on an Arduino Mega (atmega2560). The code is intended to run on a regular Arduino UNO or anything with the atmega328 chip.

That being said, I don't see why the CAN passthrough functionality wouldn't work with an ms3pro because the 29bit CAN protocol is the same as the ms2.

I haven't actually got this installed in my car yet, but when I do, I might write up a setup guide or make a quick video.

Ms3pro, 2023-05-17:
[image]
[image]

hankedan000, 2023-05-18:
If you go to File > Project Properties, and then the "CAN Devices" tab.
Do you see the EThrottle device get detected under one of the devices like this?
[image]
I have my CAN ID flash with a 5, so it's showing up as device 5 in the list.
https://github.com/hankedan000/EThrottle/blob/main/src/avr/EThrottle/can.h#L8

If it's not showing up, then something must be wrong with the CAN communication. If it does show up, then maybe you just have the device setup with the wrong CAN ID (the dialog does say "A device is responding, but not at the project assigned CAN ID.")

Ms3pro, 2023-05-18:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
Mcp2515 16mhz CAN speed 500

hankedan000, 2023-05-18:
Hmmmm you mention your MCP2515 has a 16MHz oscillator?
Most MCP2515 units I've bought off Ebay have a 8MHz oscillator. If you're is really 16MHz, then you'll need to change this constant to `MCP_16MHZ`: https://github.com/hankedan000/MegaCAN/blob/main/src/MegaCAN_Device.cpp#L99

Ms3pro, 2023-05-18:
Yes , 16mhz and I soldered this crystal myself and everything works fine on other projects with 16mhz . I 'll try to change it in the code . I removed the jumper with a knife because I was looking for a reason, now I will return the jumper to its place. I'll try again and write the results.

Ms3pro, 2023-05-18:
[image]
👍

Ms3pro, 2023-05-18:
Now I see the green CAN tx indicator, but the blue id5 is not lit. I don't understand how to configure the offset and ID of the tables. Perhaps I am incorrectly specifying the offset and id in the settings.

Ms3pro, 2023-05-18:
[image]
[image]
[image]

Ms3pro, 2023-05-18:
https://github.com/crycode-de/mcp-can-boot 
Does it need to be loaded into the arduino processor along with the throttle design?

hankedan000, 2023-05-18:
You do not need the mcp-can-boot bootloader. Although it is nice for being able to remotely flash new firmware over the CAN bus.

You do not need to enable any of the Megasquirt broadcast features that you're screenshoting for Tuner Studio to detect the CAN device.

If your CAN bus is wired correctly, and the device is functioning correctly, it should get detected and displayed in the project drop down.

At this point, you might want to put some INFO() logs in here and see if the EThrottle device is getting any CAN interrupts from messages off the bus.
[image]

I really think you might be hitting issues with trying to run this firmware on a chip that it wasn't intended for. I'm doing some interest stuff with timing the ADC measurements relative to the motor driver PWM signals. The code could be getting stuck in an ISR or something weird because it wasn't intended for an Atmega2560.

If you just want to try getting the CAN bus stuff working, I'd recommend commenting out the throttle related code and just let the CAN interrupts get serviced and the canLoop() called in the main loop.
[image]

Alternatively, if you're not interested in the EThrottle code itself, I have a much simpler GPIO CAN module example project in my MegaCAN repo (https://github.com/hankedan000/MegaCAN/tree/main/examples/GPIO). Hopefully that should run okay without having to worry the Atmega328 specific code I've developed for this project.

Ms3pro, 2023-05-21:
It turned out to connect mega 2560 GIPIO +MS3PO! Now everything works and ADC shows in the tuner studio.

Ms3pro, 2023-05-21:
[image]

hankedan000, 2023-05-21:
Awesome! Glad it works out. So it was just a wiring issue the whole time?
Did you have to comment out any of the throttle code like I was suggesting?

Ms3pro, 2023-05-21:
There were problems with the 16mhz crystal, I fixed it in the code from 8mhz to 16mhz and everything works. I don't understand how to configure tables and offsets in the settings window now, I have 6 egts and it was good if ms controlled them. Now I 'll try DBW as you said .

Ms3pro, 2023-05-21:
[image]
The throttle code does not work, I tried to comment out these lines, but it is still undetected in the device list.

hankedan000, 2023-05-21:
Hey man. I don't know what to tell you. The code is pretty robust on my setup, but there must be something different about yours. I recommend adding debug prints in the code to see where it might be getting stuck, or just not responding. You can pretty easily add a log in the CAN ISR and see if it's receiving any of those requests you have printed. If it is getting them, them follow the code and see why it's not responding to them.

The repo is not road tested at all, so I can't make any promises for how well it works yet.

hankedan000, 2023-05-26:
i've had others report that this code can communicate with an MS3 device via can pass through. closing the issue.

closed 2023-05-26 by hankedan000

Ms3pro, 2023-05-26:
Hi, your project is cool! I bought a arduino nano now everything works fine👍👏