hankedan000/megasquirt-dbw #30: ms2exstra & ms3pro ultimate no CAN connection

andreibbmwe30, 2023-05-24:
hi! an excellent project, I put together a mock-up of arduino uno + mcp2515, but the code does not want to work, it is not detected on the network. I tried our OpenGPIO-0 project.1.0 it works fine, but E Throttle doesn't want to work online. I have a question.
1) can the code work without mc33887 in can testing?
2) maybe I'm using the wrong library and that's why the code doesn't work for me, could you download all your libraries that are used in your project?
[image]
 The port monitor displays the same messages!

hankedan000, 2023-05-24:
I just tested this EThrottle code on a breadboard with just an Arduino Nano + MCP2515, and nothing else. Tuner Studio was able to detect the device just fine.

There's another guy who I've been chatting with about this similar issue where Tuner Studio says it can't communicate with the device. I'm starting to think it's related to interfacing with an MS3 ECU. For my bench testing, I'm running a DIYPNP board that's running [MS2/Extra 3.4.3 release firmware](https://www.msextra.com/downloads/archive/ms2extra-archive-firmwares/). I don't have an MS3 device to work with.

I might add some code to the MegaCAN library to dump the CAN bus traffic to help rule out any bugs. Will let you know.

hankedan000, 2023-05-25:
Okay. I've pushed up a [debug-ms3-comms-issues](https://github.com/hankedan000/MegaCAN/compare/main...debug-ms3-comms-issues) to the MegaCAN repo. If you checkout that branch and set `LOG_CAN_TRAFFIC` to 1, it will dump the CAN frames to/from the device.

Here's what the initial traffic looks like for my ms2e setup. I'd be curious to see what your looks like if it's an ms3 device.
```
INFO  | BUS >>> MCU 0x00038280 | len 4 | 80 06 00 05 00 00 00 00 | ........ | // request protocol & blocking factors
INFO  | BUS <<< MCU 0x00012830 | len 5 | 02 00 20 00 20 00 00 00 | .. . ... |
INFO  | BUS >>> MCU 0x000082f8 | len 3 | 06 00 08 00 00 00 00 00 | ........ | // request signature (20 bytes)
INFO  | BUS <<< MCU 0x00012830 | len 8 | 4f 70 65 6e 47 50 49 4f | OpenGPIO | // signature bytes 0 to 7
INFO  | BUS >>> MCU 0x002082f8 | len 3 | 06 00 08 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 8 | 2d 31 2e 30 2e 30 20 20 | -1.0.0   | // signature bytes 8 to 15
INFO  | BUS >>> MCU 0x004082f8 | len 3 | 06 00 04 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 4 | 20 20 20 00 00 00 00 00 |    ..... | // signature bytes 16 to 19
INFO  | BUS >>> MCU 0x000082f0 | len 3 | 06 00 08 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 8 | 45 54 68 72 6f 74 74 6c | EThrottl | // request version info (60 bytes)
INFO  | BUS >>> MCU 0x002082f0 | len 3 | 06 00 08 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 8 | 65 00 4f 70 65 6e 47 50 | e.OpenGP | // woops!
INFO  | BUS >>> MCU 0x004082f0 | len 3 | 06 00 08 00 00 00 00 00 | ........ | // looks like my version string isn't long enough.
INFO  | BUS <<< MCU 0x00012830 | len 8 | 49 4f 2d 31 2e 30 2e 30 | IO-1.0.0 | // doesn't seem to be an issue for me on ms2e though...
INFO  | BUS >>> MCU 0x006082f0 | len 3 | 06 00 08 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 8 | 20 20 20 20 20 00 8e ff |      ... |
INFO  | BUS >>> MCU 0x008082f0 | len 3 | 06 00 08 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 8 | ff ff ff ff ff ff ff ff | ........ |
INFO  | BUS >>> MCU 0x00a082f0 | len 3 | 06 00 08 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 8 | ff ff ff ff ff ff ff ff | ........ |
INFO  | BUS >>> MCU 0x00c082f0 | len 3 | 06 00 08 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 8 | ff ff ff ff ff ff ff ff | ........ |
INFO  | BUS >>> MCU 0x00e082f0 | len 3 | 06 00 04 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 4 | ff ff ff ff 00 00 00 00 | ........ |
INFO  | BUS >>> MCU 0x00008288 | len 3 | 06 00 08 00 00 00 00 00 | ........ | // TS starts dumping flash tables...
INFO  | BUS <<< MCU 0x00012830 | len 8 | 00 14 00 05 00 00 ff ff | ........ |
INFO  | BUS >>> MCU 0x00208288 | len 3 | 06 00 08 00 00 00 00 00 | ........ |
INFO  | BUS <<< MCU 0x00012830 | len 8 | 00 b9 03 ff 00 21 03 c6 | .....!.. |
...
```

Notice that the traffic shows my EThrottle code sending data past the end of the version string buffer. I don't think this should be happening. I should either update the length of the version request command [here](https://github.com/hankedan000/EThrottle/blob/main/src/avr/EThrottle/EThrottle-1.0.0.ini#L20), or just specific a string of length 60bytes, so I doesn't read past the end of the string like that. Regardless, I didn't think that string mattered a whole lot. Pretty sure Tuner Studio just uses it to display in the title bar.

andreibbmwe30, 2023-05-25:
Hello!  Thank you for your answer, I used MS2/Extra 3.4.4 and MS3Pro versions MS3 pre-1.6 alpha 2, I will try to do as you said, I will let you know the result later.

andreibbmwe30, 2023-05-25:
replaced the mega can library, entered what you suggested and ms2 saw the device, but ms3 does not see it.  I connected arduino + ms2extra + ms3pro ultimate and all two devices see EThrottle, that is, if there is no ms2 in the can bus, then the can bus does not work and logging does not occur.

andreibbmwe30, 2023-05-25:
[image]

andreibbmwe30, 2023-05-25:
[image]
this is if you don't run the tuner studio

andreibbmwe30, 2023-05-25:
[image]
this is if you run the tuner studio

andreibbmwe30, 2023-05-25:
thank you! everything works in ms3pro, I would like to add DBW support with base ids 256,257,260,261

hankedan000, 2023-05-25:
hmmm glad you seem to have gotten it working. do you mean it's now working with just the arduino + ms3pro (no ms2 on the bus)?

What do you mean support with base ids 256, 257, 260, 261? Are you referring to [receiving realtime broadcast data packets from the megasquirt](https://github.com/hankedan000/EThrottle/blob/main/src/avr/EThrottle/can.cpp#L164-L183)?

If you put up a pull request, I'd be happy to take a look. Note that I don't have an ms3pro to test with, so I can't promise to maintain any ms3pro specific features going forward.

andreibbmwe30, 2023-05-25:
that's right, everything works like this arduino + mp3pro, only you can't see it on the network until you add a CAN device and then it appears on the network

hankedan000, 2023-05-25:
Oh wow! I didn't know ms3 supported transmitting DBW control information like that. If you can code it up, and make it an optional feature (maybe with a #define toggle or something) then I can consider bringing it in.

hankedan000, 2023-05-26:
CAN pass through communication is functioning on MS2 and MS3 devices. Issue appears to be that the *.ini file was not loaded into the TunerStudio project.

closed 2023-05-26 by hankedan000

andreibbmwe30, 2023-05-26:
I have a piece of code for DBW, how can I add it to your project

andreibbmwe30, 2023-05-26:
#include <SPI.h>
#include <mcp2515.h>

struct can_frame canMsg;
MCP2515 mcp2515(53);

int TPS;
int sensor01=A1;
int s1, a, b;
int sensor02=A2;
int s2, e, f;

unsigned int textCounter = 0;

long unsigned int rxId;
unsigned char len = 0;
unsigned char rxBuf[8];
char msgString[128]; 

void setup() 
{
SPI.begin(); 
mcp2515.reset();
mcp2515.setBitrate(CAN_500KBPS, MCP_16MHZ);
mcp2515.setNormalMode(); 
}
void loop() 
{
if (mcp2515.readMessage(&canMsg) == MCP2515::ERROR_OK) {
for (int i = 0; i<canMsg.can_dlc; i++)   { 
}
}
switch (canMsg.can_id) {
case 260:                          // read servo position map from ms3
TPS = (float)(word(canMsg.data[0], canMsg.data[1])); TPS = (TPS / 10); //????
break;  
}  
  s1=analogRead(sensor01); //ПОЛОЖЕНИЕ PEDAL
  b=s1;               //ПОЛОЖЕНИЕ PEDAL
  s2=analogRead(sensor02); //ПОЛОЖЕНИЕ THROTTLE
  f=s2;               //ПОЛОЖЕНИЕ THROTTLE
  canMsg.can_id  = 0x100;   //MS DBW ID256       
  canMsg.can_dlc = 8;               
  canMsg.data[0]= s1 >> 8;  //ПОЛОЖЕНИЕ PEDAL             
  canMsg.data[1] = b& 0xff;         //ПОЛОЖЕНИЕ PEDAL     
  canMsg.data[2] = s2 >> 8; //ПОЛОЖЕНИЕ THROTTLE            
  canMsg.data[3] = f & 0xff;         //ПОЛОЖЕНИЕ THROTTLE
  canMsg.data[4] = 0x00;
  canMsg.data[5] = 0x00;
  canMsg.data[6] = 0x00;
  canMsg.data[7] = 0x00;
  mcp2515.sendMessage(&canMsg);
 delay(20);
 }