hankedan000/megasquirt-dbw #37: Compilation error

Sando757, 2023-09-20:
When i'm try compilate the projekt, i have a error:
Compilation error: 'virtual void EThrottleCAN::handleStandard(const uint32_t&, const uint8_t&, uint8_t*)' marked 'override', but does not override
https://i.ibb.co/PwqHggt/Screenshot-1.png

Perhaps you know something about this problem?

hankedan000, 2023-09-21:
Sorry! This project is still a work in progress.
I made an [update in my MegaCAN](https://github.com/hankedan000/MegaCAN/commit/dc590e4a4e9de5a374bc82cb9ef90117a9117a76) library that removed the usage of reference arguments in some methods.

[74b69a91bbd0d7f258c3b3d9d229f14ef7289f6f](https://github.com/hankedan000/EThrottle/pull/38/commits/74b69a91bbd0d7f258c3b3d9d229f14ef7289f6f) is the commit that updates this repo to be compliant with that change. It was only my dev branch, but I'm going to merge dev into main shortly.

hankedan000, 2023-09-21:
pull request #38 should resolve this

closed 2023-09-21 by hankedan000