hankedan000/megasquirt-dbw #39: Compilation error #2

Sando757, 2023-09-21:
Thanks for updating the code, it helped me move forward. But again there was a compiler error, as I understand it, the problem is that the <logging_impl.h> library is not connected. 

https://i.ibb.co/XLSfbJH/Screenshot-5.png

And one more problem in the archive there are no files: arduino-pid-autotuner, megaCAN and some others, I would like to help collect a complete archive to make it easier for people to start a project. If I could somehow contact you, I would send you the complete package.

hankedan000, 2023-09-21:
This is another issue with you not using the correct version of one of my libraries. In this case, I made an update in my logging library that broke compatibility. Sorry, I'm still learning how best to structure arduino libraries and projects.

I put up a branch that switches over to using submodules for all the library dependencies. Pull request #41 is open now. If you check out the submodules branch, and follow the new instructions in the README, I think it will hopefully resolve your library dependency issues.

Sando757, 2023-09-22:
It looks better. But it can still cause difficulties for some people. If possible, it would be a good idea to add a compiled ".hex" file to the archive so that it can be loaded via "Xloader", for example. This would eliminate any mistakes.

closed 2023-10-21 by hankedan000