hankedan000/megasquirt-dbw #40: Throttle body fluctuations

Sando757, 2023-09-21:
I was able to compile the firmware and load it onto the V1.1 module, connected via the CAN bus to the Megasquirt 2 board (3.4.4) - everything works. But there is a permanent driver error. The throttle body fluctuates ~5% regardless of PID settings. The motoroutPWM and current also have strong fluctuations. Tried it on 2 different boards. I don’t know what this is connected with, but perhaps these symptoms will help understand the problem. 

https://i.ibb.co/XLSfbJH/Screenshot-5.png
The screenshot shows the fluctuations, in reality they are very strong.
Perhaps I can help with something, I have megasquirt 2, megasquirt 3x and megasquirt board simulator and 5pcs psb v1.1.

hankedan000, 2023-09-22:
I'm actually struggling with jitter/stability issues myself. I think my circuit design is lacking filtering on the ADC lines. In my case, I have about a 6ft length of wires running to the throttle body, and I think there's noise coupling onto the TPS ADC wires. I've actually had the noise get so bad, that it's fried the arduino chip. I'm really struggling to resolve these issues. I've tried oscoping the lines, and I never see super high voltage spikes occur, but something is terribly wrong.

I implemented [software filtering](https://github.com/hankedan000/EThrottle/commit/49e2dc2cb0c6f079b5d8ce2d7d4c9b3abcc0ad7c) of the ADC lines, to help reduce the jitter, but it doesn't prevent the chips from eventually frying :(

Sando757, 2023-09-22:
As far as I can see, the Arduino burns out due to surges on the +12 line (VIN) from motor. There is also low-quality LM1117 stabilizer, I separated the +5 and +12 power lines, added power supply capacitors (25V 220U for both lines) and nothing else burned out. It would also be a good idea to add suppressors (SMBJ15-17) on the power lines.

About oscillations, I added filter circuits (RC), they reduced the interference in the program (graph in the PID settings), but still the throttle body has fluctuations (much less).

According to my observations, the value of motorOUTPWM changes too quickly and greatly, and there is also a rapid fluctuation of targettps + -0.5%. But I don’t know how to fix this, since I don’t understand programming well.

closed 2025-11-23 by hankedan000