hankedan000/megasquirt-dbw #43: Help Needed with Custom Bootloader and Code Upload Issue

saidsamet, 2024-11-28:
I successfully burned the bootloader using an Arduino Mega as ISP. However, when I try to upload sketches via the Arduino IDE, I receive the following error:

avrdude: stk500_recv(): programmer is not responding

Could you assist me in diagnosing and resolving this issue? You can reach me at qaftravels@gmail.com

Thank you for your support!

hankedan000, 2024-11-28:
When you say custom bootloader, are you talking about https://github.com/crycode-de/mcp-can-boot?

If so, that loader only supports CAN programming and not via USB/UART.

saidsamet, 2024-11-29:
yes, I use MCP can boot. Do you mean I have to compile the program and upload it to Arduino nano, and the change the bootloader via Arduino as ISP?

hankedan000, 2024-11-29:
No, I mean the MCP can boot bootloader doesn't support programming via the USB/UART cable. It only supports uploading programs via the can bus. To program via the USB, you'll need to put the stock Arduino bootloader back on (or use Optiboot or something else).

To flash via can bus, I use this USB to can bus adapter https://www.ebay.com/itm/404496944235?mkcid=16&mkevt=1&mkrid=711-127632-2357-0&ssspo=qcyosggntfy&sssrc=4429486&ssuid=DTkTmZSpTTK&var=&widget_ver=artemis&media=COPY. I export the Arduino sketch to a binary file via the Arduino IDE, and then program it with the command line tool that crycode-de developed.